VTAstrobotics/PHOENIX #12: Schedule meeting with Shinpaugh

VaTechAstrobotics, 2023-07-04:
Want to get his opinion on several things including:

- team structure + recruiting
- workflow
- overall team plan

ThomasGiattino, 2023-08-15:
We should add the timeline to this. I think it is important Shinpaugh reviews our deadlines and prepares for 3 design reviews in his schedule. Also I think weekly leadership meetings with Shinpaugh should be held so we need to set that up. I disagree that it's optional for this meeting.

H3nryM, 2023-08-15:
> Also I think weekly leadership meetings with Shinpaugh should be held so we need to set that up. I disagree that it's optional for this meeting.

YES CHEF 🫡

H3nryM, 2023-08-19:
https://drive.google.com/file/d/1U9No8xYJKy8rR-z4fPZ0KbyFYdSPxBJL/view?usp=drive_link

closed 2023-08-19 by H3nryM